AEFeinstein/Super-2020-Swadge-HW #21: Add some pads to circuit board

bbkiwi, 2019-07-10:
**Summary**
Make it easy to add more LEDs and other audio input. I envision doing this
to use the swadge later to make flashy things. 

- Add pads to can add more LEDs
- Add audio in pads, or plug that disconnects mic

**Technical Spec**
For more LEDs need breakout to pads of RX, +5, +3.3, GND 
(I don't think there is any standard connectors for strips (and none for rings)
Audio in: not sure exactly where to add input.

I imagine anyone want to do this would reprogram the firmware, but (at least)
for some mode, it may be possible to have NUM_LIN_LEDS be a parameter that can
be changed.

AEFeinstein, 2019-07-10:
The first thing to remember is that the Swadge is primarily targeted at normal Magfest-goers, not hardware hackers. Adding unpopulated pads to continue the LED daisy chain is easy enough. I have two concerns.

1. Do the LEDs really need both 5v and 3.3v? They're currently using only 3.3v.
2. The regulator only has [1A output](https://media.digikey.com/pdf/Data%20Sheets/Diodes%20PDFs/AP1117(I).pdf). A WS2812B can draw 50mA at full brightness. Assuming 8 LEDs onboard and no other peripherals drawing current, a user would only be able to safely add 12 more LEDs. Would it be reasonable to only have a pad for data out, and require the user to supply power to the rest of the strip?

A separate audio pad is easy enough too. A separate switching mic jack probably won't be worth the cost of the part.

1. Would you want that before the amp and filter circuitry, after, or somewhere in the middle?
2. Would it be acceptable to say "cut the trace to the current mic" to use an external mic? We can add holes too so that the cut trace would be easily reconnectable with a wire.

bbkiwi, 2019-07-11:
Hi Adam, I understand about the target audience. I'm just a 74 year old retired mathematician who likes to play with flashy lights :-) However it is fun to be able to help in someway with this even though I will never go to a Magfest. 

I am envisioning powering a longer strip with +5V out of the micro USB, although as you say the 3.3V could allow for a small number of additional lights. Thats why pads for +3.3, +5 and GND would be convienient. 

For various CC projects with the nodemcu, I usually power via +5. I also use lots of techniques mentioned
in this article:  [https://learn.adafruit.com/sipping-power-with-neopixels/insights](url) Quite bright and interesting displays using 60 LEDs can be done that draw less than 500ma. (I have on occasion blown up a regulator when my programming 'safeguards were wrong). Let the hacker beware :-)

1. Re audio, I was thinking of probably connecting an instrument pickup. I'd leave it to more experts to decide where this might be best. I'm not sure how much the gain can change with AUD_PWR. So probably somewhere in the middle. DFT32.c is sensitive to aliasing effects so any high pass filtering is helpful.  [https://github.com/cnlohr/colorchord/issues/89](url)
2. Yes I think "cutting" and "holes" would be fine.

bbkiwi, 2019-07-11:
I didn't mean to close - pushed the wrong buttom

AEFeinstein, 2019-07-12:
Retired mathematician! I never knew. And I see you have a copy of Matlab! I only have Octave :(

Exposing both 3.3V and 5V pads sounds fine. You seem to know the risks, and I think this is mostly for you, and maybe some other intrepid folk.

closed 2019-09-02 by AEFeinstein